RiS3-Lab/p2im #1: Robot(real firmware) can not be built in  System Workbench for STM32.

MaKaRuiNah, 2020-08-07:
I notice that the project has the .cfg file(such as Robot Debug.cfg).Does the Robot miss the .cfg file?

bofeng17, 2020-08-07:
I fixed this issue in https://github.com/RiS3-Lab/p2im-real_firmware/commit/32993e5a27bee2a56a7e51d498891bfcb05ed3ae

closed 2020-08-07 by bofeng17

MaKaRuiNah, 2020-08-10:
Thanks! it can be built now.Howerver,when i run the "../../../model_instantiation/fuzz.py -c fuzz.cfg", it throws the error:
"PROGRAM ABORT : Test case 'id:000000,orig:random' results in a crash"
[image]
[image]

bofeng17, 2020-08-10:
The emulator only supports 4 MCUs. You can either recompile your firmware to use one of the MCUs supported, or add your own MCU to the emulator following the instructions in https://github.com/RiS3-Lab/p2im/blob/master/docs/add_mcu.md

MaKaRuiNah, 2020-08-12:
Thanks!When i want to test a new firmware, and i use the same board and mcu as yours,it throws an error,how can i solve it?
[image]

bofeng17, 2020-08-12:
The firmware hangs when running with the seed input. You may want to dump the execution trace to see where the firmware hangs. 

Please append `-d exec,int -D qemu.log` option to the command below and run it. 
[image]
The firmware execution trace is dumped to `qemu.log`.

MaKaRuiNah, 2020-08-14:
Thank you very much! I want to know where the code crashes, i run the "run_fw.py".however, i only get the address(like 0x08003658),can i get the line number where the code crashes?
[image]