RiS3-Lab/p2im #6: How to compile using ARMCC

JerryWang304, 2020-10-21:
I use Keil v5, which uses ARMCC (ARM compiler ) to compile. It issues errors because of assembly codes
```
..\Src\afl_call.c(5): warning:  #1207-D: unknown attribute "naked"
  __attribute__ ((naked)) uint32_t aflCall(__attribute__ ((unused)) uint32_t a0, __attribute__ ((unused)) uint32_t a1, __attribute__ ((unused)) int32_t a2) {
..\Src\afl_call.c(11): error:  #130: expected a "{"
      __asm__ __volatile__ ("svc $0x3f\n\t"
..\Src\afl_call.c(11): warning:  #174-D: expression has no effect
      __asm__ __volatile__ ("svc $0x3f\n\t"
..\Src\afl_call.c(13): warning:  #940-D: missing return statement at end of non-void function "aflCall"
  }
..\Src\afl_call.c: 3 warnings, 1 error
```
So I change the assembly code to :
```
   __asm("svc $0x3f\n\t"
           "bx %lr\n\t");
```
Keil also failed to compile:
```
..\Src\afl_call.c(5): warning:  #1207-D: unknown attribute "naked"
  __attribute__ ((naked)) uint32_t aflCall(__attribute__ ((unused)) uint32_t a0, __attribute__ ((unused)) uint32_t a1, __attribute__ ((unused)) int32_t a2) {
..\Src\afl_call.c(11): error:  #20: identifier "$0x3f" is undefined
      __asm("svc $0x3f\n\t"
..\Src\afl_call.c(13): warning:  #940-D: missing return statement at end of non-void function "aflCall"
  }
..\Src\afl_call.c: 2 warnings, 1 error
``` 
So, is there any way to generate required firmware for fuzzing using ARMCC ?

bofeng17, 2020-10-22:
I didn't try to compile firmware by ARMCC. From the error message, `..\Src\afl_call.c(11): error:  #20: identifier "$0x3f" is undefined`, it seems ARMCC has a different syntax for inline assembly from GCC. "$0x3f" is a constant that ARMCC fails to recognize.

JerryWang304, 2020-10-22:
Now I compile my project successfully. But AFL failed to start because `All test cases time out, giving up! `. And p2im prints `qemu hangs(pid: 26155). seed input should not hang!`
Have you ever met this error before ? Any idea to solve it ?

bofeng17, 2020-11-27:
A longer AFL timeout value (in milliseconds) may help https://github.com/RiS3-Lab/p2im/blob/master/fuzzing/templates/fuzz.cfg.template#L29

closed 2020-11-29 by bofeng17

dokodokoR, 2021-11-14:
> I use Keil v5, which uses ARMCC (ARM compiler ) to compile. It issues errors because of assembly codes
>
> ```
> ..\Src\afl_call.c(5): warning:  #1207-D: unknown attribute "naked"
>   __attribute__ [...]

dokodokoR, 2021-11-14:
@JerryWang304 

> Now I compile my project successfully. But AFL failed to start because `All test cases time out, giving up! `. And p2im prints `qemu hangs(pid: 26155). seed input should not hang!` Have you ever met this error before ? Any idea to solve it ?

How to change the assembly code? What else have you changed in the code below?

> ```
>    __asm("svc $0x3f\n\t"
>            "bx %lr\n\t");
> ```