RiS3-Lab/p2im #9: QEMU image librt: undefined symbol: __clock_nanosleep

andreia-oca, 2020-12-15:
I am trying to use the precompiled QEMU image to fuzz with P2IM and I am receiving the following error:
```
p2im/qemu/precompiled_bin/qemu-system-gnuarmeclipse: symbol lookup error: /home/circleci/p2im/qemu/precompiled_bin/librt.so.1: undefined symbol: __clock_nanosleep, version GLIBC_PRIVATE
```
I tried to recompile using the [instructions](https://github.com/RiS3-Lab/p2im/blob/master/docs/build_qemu.md) from the repo, but the error persisted.

Unfortunately, fuzz.py and me.py are also failing due to this error.

Does anyone have a clue how to solve it?

AustinParkes, 2020-12-15:
Are you using 64-bit Ubuntu 16.04.? I had unexplained issues with the wrong version.

alejoseb, 2020-12-15:
Are you using Ubuntu 20.04?  we are aware of some issues with it. Please use 16.04 or 18.4 LTS.

closed 2020-12-23 by bofeng17

andreia-oca, 2021-01-03:
You were right. Using Ubuntu 20.04 was the cause. I switched to Ubuntu 16.04 and now the QEMU image works fine. 
Thank you!