RiS3-Lab/p2im #11: me.py error "Stage 1 returned due to unexpected reasons!"

andreia-oca, 2021-01-03:
I am trying to run P2IM with a precompiled firmware from the repo (the one named "Console"). I have followed the steps from the README and the command `model_instantiation/fuzz.py -c $WORKING_DIR/fuzz.cfg` is triggering the following error:
```
cmd_me0: /root/p2im//model_instantiation/me.py -c /root/p2im/fuzzing/console/01/fuzz.cfg --run-num 0 --print-to-file

Change working dir to: 0/
CWD: /root/p2im/fuzzing/console/01/0
Redirect stdout to file named stdout
Traceback (most recent call last):
  File "/root/p2im//model_instantiation/me.py", line 1146, in <module>
    ret_val = stage1()
  File "/root/p2im//model_instantiation/me.py", line 347, in stage1
    ret_val = qemu_run(cmd, cfg.retry_num, stage)
  File "/root/p2im//model_instantiation/me.py", line 188, in qemu_run
    color_print(error_rv[stage][ret_val], "red")
KeyError: 1
``` 

Do you know what could be the cause for this?

andreia-oca, 2021-01-03:
In order to reproduce the error easier, I have created a docker image that installs and compiles all the dependencies for P2IM and also runs it on the firmware "Console". You can access it in this forked [repo](https://github.com/andreia-oca/p2im/tree/master/docker).

To recreate the error, simply run:
```
cd /path/to/p2im/repo/docker
make test
```

If you find convenient, I can also create a pull request to make the Dockerfile widely available in the your original repository.

bofeng17, 2021-01-04:
Hi Andreia, can you attach this file `/root/p2im/fuzzing/console/01/0/stdout` so that I can figure out what causes stage 1 to return?

andreia-oca, 2021-01-04:
The contents of `/root/p2im/fuzzing/console/01/0/stdout` are:
```
cmd to launch this script: /root/p2im//model_instantiation/me.py -c /root/p2im/fuzzing/console/01/fuzz.cfg --run-num 0 --print-to-file

args after processing: Namespace(afl_file=None, config='/root/p2im/fuzzing/console/01/fuzz.cfg', eval=False, gt=None, model_if=None, print_to_file=True, run_from_fs=False, run_num='0')

configurations after processing: Namespace(board='NUCLEO-F103RB', img='/root/p2im/fuzzing/console/01/Console', log_f='/root/p2im/fuzzing/console/01/me.log', mcu='STM32F103RB', objdump='/root/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-objdump', peri_addr_range=512, qemu_bin='/root/p2im/qemu/precompiled_bin/qemu-system-gnuarmeclipse', qemu_log='unimp,guest_errors,int', retry_num=3)

depth 1, stage: SR_R_ID
cmd: /root/p2im/qemu/precompiled_bin/qemu-system-gnuarmeclipse -verbose -verbose -d unimp,guest_errors,int -nographic -board NUCLEO-F103RB -mcu STM32F103RB -image /root/p2im/fuzzing/console/01/Console -pm-stage 1 -trace trace-depth:1,stage:1.0 -reg-acc reg_acc-depth:1,stage:1.0 -model-output model-depth:1,stage:1.0.json
ret_val: 0xff
ret_val == 0xff, re-run it!
ret_val: 0xff
ret_val == 0xff, re-run it!
ret_val: 0xff
ret_val == 0xff, re-run it!

exit_callback is invoked

Execution time(seconds): 
0.03566741943359375
```

I tried to run `/root/p2im/qemu/precompiled_bin/qemu-system-gnuarmeclipse -verbose -verbose -d unimp,guest_errors,int -nographic -board NUCLEO-F103RB -mcu STM32F103RB -image /root/p2im/fuzzing/console/01/Console -pm-stage 1 -trace trace-depth:1,stage:1.0 -reg-acc reg_acc-depth:1,stage:1.0 -model-output model-depth:1,stage:1.0.json
`  as a standalone command and its output is:
```
GNU ARM Eclipse 64-bits QEMU v2.3.50 (qemu-system-gnuarmeclipse).

(process:59): GLib-WARNING **: /Host/Work/qemu/glib-2.51.0/glib/gmem.c:483: custom memory allocation vtable not supported
Board: 'NUCLEO-F103RB' (ST Nucleo Development Board for STM32 F1 series).
Device: 'STM32F103RB' (Cortex-M3 r0p1, MPU), Flash: 128 kB, RAM: 20 kB.
Image: '/root/p2im/fuzzing/console/01/Console'.
Command line: (none).
[0, 0]   1-th(total   1-th) 	unassigned mem_r *0x0
[0, 0]   2-th(total   2-th) 	unassigned mem_r *0x4
Load   1024 bytes at 0x00000000-0x000003FF.
Load     16 bytes at 0x00000400-0x0000040F.
Load  30800 bytes at 0x00000410-0x00007C5F.
Load    336 bytes at 0x00007C60-0x00007DAF.
Load      0 bytes at 0x1FFF0000-0x1FFEFFFF.
Cortex-M3 r0p1 core initialised.
QEMU 2.3.50 monitor - type 'help' for more information
(qemu) QEMU 2.3.50 monitor - type 'help' for more information
(qemu) Cortex-M3 r0p1 core reset.
[0, 0] illegal read at 0x99c
```
Also, I did not mention previously, I am testing commands in a docker and also inside a virtual machine using VMWare with Ubuntu 16.04.

bofeng17, 2021-01-04:
The console firmware is based on MK64FN1M0VLL12 mcu. Can you try to set this mcu/board in `fuzz.cfg` https://github.com/RiS3-Lab/p2im/blob/master/fuzzing/templates/fuzz.cfg.template#L56-L57?

andreia-oca, 2021-01-04:
This was the problem. I used the mcu/board that you've indicated and now everything works fine. Thank you!

A small following question. How do I know on which mcu are the other real-world firmwares in this repo based?

bofeng17, 2021-01-04:
I just added this missing information to readme https://github.com/RiS3-Lab/p2im#preparing-the-configuration-file

andreia-oca, 2021-01-04:
Perfect. Thanks a lot!

closed 2021-01-04 by andreia-oca