RiS3-Lab/p2im #13: KeyError:255, QEMU's return value is not found

youngwoo013, 2021-04-17:
Hello, I am studying and experimenting on P2IM.
I want to run firmware using P2IM.The error "KeyError:255" occurs in the me.py code. This means that the return value will be 0xff when QEMU is run as subprocess and it is not defined as the rv dictionary of me.py code. This results in an error.
**What is the meaning of the return value of the QEMU, 0xff?**

[image]

If I ignore this error and proceed, **there is a myriad of rounds with the phrase "There is aup, run Me."** (where the "Stage 2 returned to unexpected reasons!" part indicates a KeyError:255 error, and simply added 0xff to the dictionary not to print errors.)

[image]

And at the end, fuzzing does not proceed with the following error.
**I proceeded with the firmware that added the mcu and startForkserver part, and I wonder how to solve the error.**

[image]

bofeng17, 2021-04-21:
Return value 0xff means qemu detects an illegal memory read/write which is outside ram, flash, or peripheral regions. The check is performed at https://github.com/RiS3-Lab/p2im/blob/master/qemu/src/qemu.git/softmmu_template.h#L186-L197 and  https://github.com/RiS3-Lab/p2im/blob/master/qemu/src/qemu.git/softmmu_template.h#L414-L423. 

To figure out which instruction in firmware triggered the illegal access, please run the qemu command printed right above your first screenshot in your shell.

youngwoo013, 2021-05-18:
I'm sorry for the late reply. I couldn't solve the problem, but it helped me a lot to figure it out. Thank you.

closed 2021-05-19 by bofeng17