RiS3-Lab/p2im #14: Encounter a problem: get qemu log when fuzzing and then can not find crash anymore

liuliqaz, 2021-04-30:
Hello, I change function "get_qemu_arg()" like follow, try to get firmware running log when fuzzing. After change, I can not get any crash in fuzzing. If I change back, I can find a crash.

[image]

Thanks for your help!!!

dokodokoR, 2021-12-28:
@liuliqaz 
The execution time may have been increased by the log, and it may have been determined to be a "outputs/hang".

bofeng17, 2022-11-22:
I agree with @dokodokoR that dumping log during fuzzing is very slow and are likely to cause hangs during fuzzing. You can rerun the fuzzer-generated test cases offline to dump log.

closed 2022-11-22 by bofeng17