RiS3-Lab/p2im #19: Memory limit

soemthlng, 2022-09-29:
I am using p2im in Ubuntu 16.05 LTS on VMWare.
The settings of VM are as below.
`Memory    14.9GB
Processors    4
Hard Disk    60GB`
I used PLC from `https://github.com/RiS3-Lab/p2im-real_firmware/tree/master/PLC` and run with this code `<repo_path>/model_instantiation/fuzz.py -c fuzz.cfg`

This error occurs.

    [-] Oops, the program crashed with one of the test cases provided. There are
    several possible explanations:

    - The test case causes known crashes under normal working conditions. If
      so, please remove it. The fuzzer should be seeded with interesting
      inputs - but not ones that cause an outright crash.

    - The current memory limit (2.00 GB) is too low for this program, causing
      it to die due to OOM when parsing valid files. To fix this, try
      bumping it up with the -m setting in the command line. If in doubt,
      try something along the lines of:

      ( ulimit -Sv $[2047 << 10]; /path/to/binary [...] <testcase )

      Tip: you can use http://jwilk.net/software/recidivm to quickly
      estimate the required amount of virtual memory for the binary. Also,
      if you are using ASAN, see docs/notes_for_asan.txt.

    - Least likely, there is a horrible bug in the fuzzer. If other options
      fail, poke <lcamtuf@coredump.cx> for troubleshooting tips.

    [-] PROGRAM ABORT : Test case 'id:000000,orig:random' results in a crash
         Location : perform_dry_run(), afl-fuzz.c:2781

To solve this problem, I typed this command
`    root@ubuntu:/home/aiden/p2im/fuzzing/PLC/5# ulimit -Sv $[2047 << 10]; /home/aiden/p2im/fuzzing/PLC/5/PLC.elf`
Also this error occurs    
`    bash: /home/aiden/p2im/fuzzing/PLC/5/PLC.elf: cannot execute binary file: Exec format error`

Is there any way to make p2im recognize the 16 GB of memory allocated to the VM? or is there no other way to unlock the limit via ulimit -Sv? 
/etc/security/limits.conf has already been modified.

bofeng17, 2022-11-22:
It's most likely not a memory limit issue. I guess it's caused by "test cases crash", which causes QEMU crash or exit with a non-zero value (treated as a crash), you can figure out the "crash" reason by `./run_fw.py` following the instructions in https://github.com/RiS3-Lab/p2im#analyzing-crashinghanging-input

jackniu1234, 2023-02-20:
I run ./run_fw.py mentioned by @bofeng17 and I figured out that the "crash" is caused by the symbol link in the precompiled qemu provided by the p2im corrupted!

I guess the reason is that I copied the git repository from windows to ubuntu virtual machine.

One solution I found is listed in 
https://askubuntu.com/questions/24266/error-while-loading-shared-libraries-file-too-short

closed 2023-10-02 by bofeng17