RiS3-Lab/p2im #23: Docker Build Script Failed To Fetch

Tyro533, 2023-08-29:
I've followed the instructions in build_qemu.md but once i execute the 
`WORK_FOLDER_PATH=`pwd`/src ./build_scripts/build-qemu.sh --deb64 --no-strip`
command i end up with the following output. I've seen other issues with similar docker problems that suggest running an update/upgrade in the container but it's not like i can open a shell in the container since it starts up executing that build script right away. Is there a solution?

```
Using "/media/sf_Masterarbeit/DICE-DMA-Emulation/p2im/qemu/src" as Work folder...
Helper script: "/media/sf_Masterarbeit/DICE-DMA-Emulation/p2im/qemu/src/scripts/build-helper.sh".
Script "./build_scripts/build-qemu.sh" started at Tue Aug 29 06:30:56 PM EDT 2023.

Running on Ubuntu 64-bits.

Checking host curl...
curl 7.85.0 (x86_64-pc-linux-gnu) libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.49.0 librtmp/2.3

Checking host git...
git version 2.37.2

Checking Docker...
Docker version 24.0.5, build ced0996

Checking host automake...
automake (GNU automake) 1.16.5

Checking host patch...
GNU patch 2.7.6

Checking host tar...
tar (GNU tar) 1.34
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.

Checking host unzip...
UnZip 6.00 of 20 April 2009, by Debian. Original by Info-ZIP.

Checking host makeinfo...
texi2any (GNU texinfo) 6.8

================================================================================
Creating Debian 64-bits archive...

Running "build.sh" script inside "qemu-debian64-build" container, image "ilegeul/debian:8-gnuarm-gcc-x11-v4"...

Linux docker 5.19.0-46-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun 16 13:30:11 UTC 2023 x86_64 GNU/Linux

Checking automake...
automake (GNU automake) 1.14.1
Checking pkg-config...
0.28
Checking readelf...
GNU readelf (GNU Binutils for Debian) 2.25
Checking gcc...
gcc (Debian 4.9.2-10) 4.9.2
Checking patchelf...
patchelf 0.8
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libblkid-dev libdrm-dev uuid-dev
The following NEW packages will be installed:
  libblkid-dev libdrm-dev libmount-dev mesa-common-dev uuid-dev
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 939 kB of archives.
After this operation, 4548 kB of additional disk space will be used.
Err http://http.debian.net/debian/ jessie/main uuid-dev amd64 2.25.2-6
  404  Not Found
Err http://http.debian.net/debian/ jessie/main libblkid-dev amd64 2.25.2-6
  404  Not Found
Err http://http.debian.net/debian/ jessie/main libdrm-dev amd64 2.4.58-2
  404  Not Found
Err http://http.debian.net/debian/ jessie/main mesa-common-dev amd64 10.3.2-1+deb8u1
  404  Not Found
Err http://http.debian.net/debian/ jessie/main libmount-dev amd64 2.25.2-6
  404  Not Found
E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/uuid-dev_2.25.2-6_amd64.deb  404  Not Found

E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/libblkid-dev_2.25.2-6_amd64.deb  404  Not Found

E: Failed to fetch http://http.debian.net/debian/pool/main/libd/libdrm/libdrm-dev_2.4.58-2_amd64.deb  404  Not Found

E: Failed to fetch http://http.debian.net/debian/pool/main/m/mesa/mesa-common-dev_10.3.2-1+deb8u1_amd64.deb  404  Not Found

E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/libmount-dev_2.25.2-6_amd64.deb  404  Not Found

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

xgandiaga, 2023-09-01:
Trying to find a fix to this too. Debian Jessie images are deprecated, that is why the build script within the Docker container fails.
However, that is out of the scope for P2IM. It is an error linked to the debian8 image used to build gnuarmeclipse qemu

> I've followed the instructions in build_qemu.md but once i execute the `WORK_FOLDER_PATH=`pwd`/src ./build_scripts/build-qemu.sh --deb64 --no-strip` command i end up with the following output. I've [...]

Trying to fix it from my part too. 
The image used to build QEMU within the docker container is a debian 8/Jessie. 
Debian repos for Jessie do not exist at debian.net. Thus the error.
The easy fix would be to change the build environment within  ...../build_scripts/build-qemu.sh
That probably means creating a debian based image that has the requirements for building QEMU gnuarmeclipse
- gcc toolchain
- other dependencies (not gonna write them out, they are at https://wiki.qemu.org/Hosts/Linux#Building_QEMU_for_Linux)

xgandiaga, 2023-09-01:
> > I've followed the instructions in build_qemu.md but once i execute the `WORK_FOLDER_PATH=`pwd`/src ./build_scripts/build-qemu.sh --deb64 --no-strip` command i end up with the following output. I've seen other issues with similar docker problems that suggest running an update/upgrade in the container but it's not like i can open a shell in the container since it starts up executing that build script right away. Is there a solution?
> > ```
> > Using "/media/sf_Masterarbeit/DICE-DMA-Emulation/p2im/qemu/src" as Work folder...
> > Helper script: "/media/sf_Masterarbeit/DICE-DMA-Emulation/p2im/qemu/src/scripts/build-helper.sh".
> > Script "./build_scripts/build-qemu.sh" started at Tue Aug 29 06:30:56 PM EDT 2023.
> > 
> > Running on Ubuntu 64-bits.
> > 
> > Checking host curl...
> > curl 7.85.0 (x86_64-pc-linux-gnu) libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.49.0 librtmp/2.3
> > 
> > Checking host git...
> > git version 2.37.2
> > 
> > Checking Docker...
> > Docker version 24.0.5, build ced0996
> > 
> > Checking host automake...
> > automake (GNU automake) 1.16.5
> > 
> > Checking host patch...
> > GNU patch 2.7.6
> > 
> > Checking host tar...
> > tar (GNU tar) 1.34
> > Copyright (C) 2021 Free Software Foundation, Inc.
> > License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
> > This is free software: you are free to change and redistribute it.
> > There is NO WARRANTY, to the extent permitted by law.
> > 
> > Written by John Gilmore and Jay Fenlason.
> > 
> > Checking host unzip...
> > UnZip 6.00 of 20 April 2009, by Debian. Original by Info-ZIP.
> > 
> > Checking host makeinfo...
> > texi2any (GNU texinfo) 6.8
> > 
> > ================================================================================
> > Creating Debian 64-bits archive...
> > 
> > Running "build.sh" script inside "qemu-debian64-build" container, image "ilegeul/debian:8-gnuarm-gcc-x11-v4"...
> > 
> > Linux docker 5.19.0-46-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun 16 13:30:11 UTC 2023 x86_64 GNU/Linux
> > 
> > Checking automake...
> > automake (GNU automake) 1.14.1
> > Checking pkg-config...
> > 0.28
> > Checking readelf...
> > GNU readelf (GNU Binutils for Debian) 2.25
> > Checking gcc...
> > gcc (Debian 4.9.2-10) 4.9.2
> > Checking patchelf...
> > patchelf 0.8
> > Reading package lists... Done
> > Building dependency tree       
> > Reading state information... Done
> > The following extra packages will be installed:
> >   libblkid-dev libdrm-dev uuid-dev
> > The following NEW packages will be installed:
> >   libblkid-dev libdrm-dev libmount-dev mesa-common-dev uuid-dev
> > 0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
> > Need to get 939 kB of archives.
> > After this operation, 4548 kB of additional disk space will be used.
> > Err http://http.debian.net/debian/ jessie/main uuid-dev amd64 2.25.2-6
> >   404  Not Found
> > Err http://http.debian.net/debian/ jessie/main libblkid-dev amd64 2.25.2-6
> >   404  Not Found
> > Err http://http.debian.net/debian/ jessie/main libdrm-dev amd64 2.4.58-2
> >   404  Not Found
> > Err http://http.debian.net/debian/ jessie/main mesa-common-dev amd64 10.3.2-1+deb8u1
> >   404  Not Found
> > Err http://http.debian.net/debian/ jessie/main libmount-dev amd64 2.25.2-6
> >   404  Not Found
> > E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/uuid-dev_2.25.2-6_amd64.deb  404  Not Found
> > 
> > E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/libblkid-dev_2.25.2-6_amd64.deb  404  Not Found
> > 
> > E: Failed to fetch http://http.debian.net/debian/pool/main/libd/libdrm/libdrm-dev_2.4.58-2_amd64.deb  404  Not Found
> > 
> > E: Failed to fetch http://http.debian.net/debian/pool/main/m/mesa/mesa-common-dev_10.3.2-1+deb8u1_amd64.deb  404  Not Found
> > 
> > E: Failed to fetch http://http.debian.net/debian/pool/main/u/util-linux/libmount-dev_2.25.2-6_amd64.deb  404  Not Found
> > 
> > E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
> > ```
> 
> Trying to fix it from my part too. The image used to build QEMU within the docker container is a debian 8/Jessie. Debian repos for Jessie do not exist at debian.net. Thus the error. The easy fix would be to change the build environment within ...../build_scripts/build-qemu.sh That probably means creating a debian based docker image that has the requirements for building QEMU gnuarmeclipse
> 
>     * gcc toolchain
> 
>     * other dependencies (not gonna write them out, they are at https://wiki.qemu.org/Hosts/Linux#Building_QEMU_for_Linux)
> 
> 
> OPTION B is to build it in your machine by installing all dependencies listed in https://wiki.qemu.org/Hosts/Linux#Building_QEMU_for_Linux

Quick comment on OPTION B

Compiling manually only works if you can set it up so gnuarmeclipse can a target when executing 
`./configure --target-list=gnuarmeclipse --python=PATHTOPYTHON2.x`

So probably fixing OPTION A is preferable.

Right now those docker containers are no longer maintained and the original dev is working on QEMU xPack.
So, the options are

OPTION A
Create a new environment/docker container with debian 8/Jessie
(Employing archive.debian.org as a source so apt update does not crash)

OPTION B
Probably discarded

OPTION C
Try to find a way of building P2IM's QEMU through xPack

Tyro533, 2023-09-10:
I see. Thank you for the pointers! If you manage to fix it, could you let me know? I'm of course going to try to do it myself but my docker/linux knowledge is limited so it might take a while

xgandiaga, 2023-09-11:
> I see. Thank you for the pointers! If you manage to fix it, could you let me know? I'm of course going to try to do it myself but my docker/linux knowledge is limited so it might take a while

Got the fixes up in my github account
https://github.com/xgandiaga/DRIVERS/tree/main

I made the build scripts point to new docker images I created in my dockerhub account.
Those point to archive.debian.org instead of deb.debian.org
Thus the apt updates don't crash
And the build goes through

(I'll fix the Licensing issues when I can. I kinda conjured up a fix from thin air quickly without looking to the licenses)

alejoseb, 2023-09-11:
Hi @xgandiaga and @valuta1995, please consider submitting a pull request with the solution to the issue.